stevieb9/wiringpi-api #8: Build failure

jberger, 2017-01-14:
Hi, I'm having a build failure trying to install the module (installed as a prereq dependency of installing `RPi::WiringPi`). I really have no idea what is wrong, but I'm happy to help debug if necessary (or maybe this is something that's known and/or PEBKAC). For faster debugging I'm almost always on irc as jberger on irc.perl.org.

The output of `gpio -v` is

```
gpio version: 2.36
Copyright (c) 2012-2015 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Embest 
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.
```

The build output is

```
cpanm (App::cpanminus) 1.7042 on perl 5.020002 built for arm-linux-gnueabihf-thread-multi-64int
Work directory is /home/joel/.cpanm/work/1484368532.16220
You have make /usr/bin/make
You have /usr/bin/wget
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
Searching RPi::WiringPi () on cpanmetadb ...
--> Working on RPi::WiringPi
Fetching http://www.cpan.org/authors/id/S/ST/STEVEB/RPi-WiringPi-1.02.tar.gz
-> OK
Unpacking RPi-WiringPi-1.02.tar.gz
Entering RPi-WiringPi-1.02
Checking configure dependencies from META.json
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Configuring RPi-WiringPi-1.02
Running Makefile.PL
Warning: prerequisite WiringPi::API 1.04 not found.
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for RPi::WiringPi
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Checking if you have WiringPi::API 1.04 ... No
Checking if you have RPi::WiringPi::Constant 0.02 ... Yes (0.02)
==> Found dependencies: WiringPi::API
Searching WiringPi::API (1.04) on cpanmetadb ...
--> Working on WiringPi::API
Fetching http://www.cpan.org/authors/id/S/ST/STEVEB/WiringPi-API-1.05.tar.gz
-> OK
Unpacking WiringPi-API-1.05.tar.gz
Entering WiringPi-API-1.05
Checking configure dependencies from META.json
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Configuring WiringPi-API-1.05
Running Makefile.PL
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for WiringPi::API
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Building and testing WiringPi-API-1.05
cp lib/WiringPi/API.pm blib/lib/WiringPi/API.pm
Running Mkbootstrap for API ()
chmod 644 "API.bs"
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- API.bs blib/arch/auto/WiringPi/API/API.bs 644
"/usr/bin/perl" "/usr/share/perl/5.20/ExtUtils/xsubpp"  -typemap '/usr/share/perl/5.20/ExtUtils/typemap' -typemap '/home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/typemap'  API.xs > API.xsc
Please specify prototyping behavior for API.xs (see perlxs manual)
mv API.xsc API.c
cc -c  -I. -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"1.05\" -DXS_VERSION=\"1.05\" -fPIC "-I/usr/lib/arm-linux-gnueabihf/perl/5.20/CORE"   API.c
rm -f blib/arch/auto/WiringPi/API/API.so
LD_RUN_PATH="/usr/local/lib" cc  -shared -L/usr/local/lib -fstack-protector API.o  -o blib/arch/auto/WiringPi/API/API.so  \
   -lwiringPi -lwiringPiDev   \
  
chmod 755 blib/arch/auto/WiringPi/API/API.so
Manifying 1 pod document
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- API.bs blib/arch/auto/WiringPi/API/API.bs 644
PERL_DL_NONLAZY=1 "/usr/bin/perl" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib/lib', 'blib/arch')" t/*.t

#   Failed test 'use WiringPi::API;'
#   at t/00-load.t line 12.
#     Tried to use 'WiringPi::API'.
#     Error:  Can't load '/home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so' for module WiringPi::API: /home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so: undefined symbol: piBoardRev at /usr/lib/arm-linux-gnueabihf/perl/5.20/DynaLoader.pm line 187.
#  at t/00-load.t line 12.
# Compilation failed in require at t/00-load.t line 12.
# BEGIN failed--compilation aborted at t/00-load.t line 12.
# Looks like you failed 1 test of 1.
t/00-load.t .......... 
Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/1 subtests 

#   Failed test 'use WiringPi::API;'
#   at t/05-phys_to_wpi.t line 6.
#     Tried to use 'WiringPi::API'.
#     Error:  Can't load '/home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so' for module WiringPi::API: /home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so: undefined symbol: piBoardRev at /usr/lib/arm-linux-gnueabihf/perl/5.20/DynaLoader.pm line 187.
#  at t/05-phys_to_wpi.t line 6.
# Compilation failed in require at t/05-phys_to_wpi.t line 6.
# BEGIN failed--compilation aborted at t/05-phys_to_wpi.t line 6.
Undefined subroutine &WiringPi::API::physPinToWpi called at /home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/lib/WiringPi/API.pm line 145.
# Tests were run but no plan was declared and done_testing() was not seen.
# Looks like your test exited with 255 just after 1.
t/05-phys_to_wpi.t ... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 1/1 subtests 
Can't load '/home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so' for module WiringPi::API: /home/joel/.cpanm/work/1484368532.16220/WiringPi-API-1.05/blib/arch/auto/WiringPi/API/API.so: undefined symbol: piBoardRev at /usr/lib/arm-linux-gnueabihf/perl/5.20/DynaLoader.pm line 187.
 at t/10-phys-to-gpio.t line 5.
Compilation failed in require at t/10-phys-to-gpio.t line 5.
BEGIN failed--compilation aborted at t/10-phys-to-gpio.t line 5.
t/10-phys-to-gpio.t .. 
Dubious, test returned 255 (wstat 65280, 0xff00)
No subtests run 

Test Summary Report
-------------------
t/00-load.t        (Wstat: 256 Tests: 1 Failed: 1)
  Failed test:  1
  Non-zero exit status: 1
t/05-phys_to_wpi.t (Wstat: 65280 Tests: 1 Failed: 1)
  Failed test:  1
  Non-zero exit status: 255
  Parse errors: No plan found in TAP output
t/10-phys-to-gpio.t (Wstat: 65280 Tests: 0 Failed: 0)
  Non-zero exit status: 255
  Parse errors: No plan found in TAP output
Files=3, Tests=2,  1 wallclock secs ( 0.10 usr  0.01 sys +  0.36 cusr  0.02 csys =  0.49 CPU)
Result: FAIL
Failed 3/3 test programs. 2/2 subtests failed.
Makefile:1029: recipe for target 'test_dynamic' failed
make: *** [test_dynamic] Error 255
-> FAIL Installing WiringPi::API failed. See /home/joel/.cpanm/work/1484368532.16220/build.log for details. Retry with --force to force install it.
-> FAIL Installing the dependencies failed: Module 'WiringPi::API' is not installed
-> FAIL Bailing out the installation for RPi-WiringPi-1.02.
```

stevieb9, 2017-01-14:
Hi Joel, thanks much for the report!

Yes, this came to be a known issue, and I've corrected it in the most recent commit to WiringPi::API (the core that RPi::WiringPi requires), pushed it to the CPAN. I will be updating RPi::WiringPi later today with a new version.

What happened was when Gordon updated the C wiringPi libraries, he removed the piBoardRev() function and replaced it with piGpioLayout(). The symbol errors you're receiving are saying WiringPi::API can't find the former.

Also, I have added in new functionality of the core, and this new functionality will be represented in RPi::WiringPi later today:

- access to SR74HC595 shift registers (four total, for a max total of 32 new outputs)
- access to the BMP180 barometric pressure/temperature sensor
- added ability to read and write to analog pseudo pins

I've also written a bunch of individual distributions for the ADS1xxx series analog to digital converters etc, that I will eventually incorporate into the base core of this dist.

I'll comment and close this ticket when I get RPi::WiringPi fully updated this afternoon.

stevieb9, 2017-01-14:
Darn... I missed the `physPinToWpi()` symbol. I'll have to take care of that too in this distribution. I'll accomplish that today as well.

stevieb9, 2017-01-14:
The `physPinToWpi()` issue is now fixed in v2.36.1. Again, I'll close this when RPi::WiringPi is complete

stevieb9, 2017-01-14:
I have committed and pushed to the CPAN RPi::WiringPi v2.36.2 (skip over the .1 version as I forgot to update the WiringPi::API prereq version).

There are a few issues in the POD I need to clean up, and I need to set back up my unit test configuration tomorrow to ensure all is working as expected (the majority is as I tested manually). If you find anything, please let me know, but I will be doing another update tomorrow to cover all known items after fully testing.

Thanks again!

jberger, 2017-01-14:
Cheers. Thanks for the quick work! I'm not near my Pi tonight or I'd give
it a try. Likely I will tomorrow!

stevieb9, 2017-01-15:
Feel free to reach out directly steveb@cpan.org if you have any comments or
questions. I'll be glad to get any feedback to make the code better and
especially make the documentation easier to follow.

I'll be setting up a permanent physical test layout today so that I can
just plug in a Pi, and run all tests. These RPi distributions have far less
test code than I always write, but to be honest, that's because it takes a
long time to physically change things up for testing. I have a few digital
potentiometers on the way, so once those arrive, I'll be able to have
everything automated; signals go out that I can regulate, then draw them
back in either digitally or analog to read the values. Once I get the new
pieces, I will add tests for everything (get near 100% coverage like my
other dists) and be able to test with a single command with no physical
interaction.

Currently, I test as I go, which clearly isn't optimal.

esc120, 2017-02-12:
Hi, I have a build failure when installing WiringPi::API
Here is my log:

```
pi@raspberrypi:~ $ sudo cpan install WiringPi::API
Loading internal null logger. Install Log::Log4perl for logging messages
Reading '/root/.cpan/Metadata'
  Database was generated on Sun, 12 Feb 2017 14:41:05 GMT
Running install for module 'WiringPi::API'
Checksum for /root/.cpan/sources/authors/id/S/ST/STEVEB/WiringPi-API-2.36.4.tar.gz ok
'YAML' not installed, will not store persistent state
Configuring S/ST/STEVEB/WiringPi-API-2.36.4.tar.gz with Makefile.PL
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for WiringPi::API
Writing MYMETA.yml and MYMETA.json
  STEVEB/WiringPi-API-2.36.4.tar.gz
  /usr/bin/perl Makefile.PL INSTALLDIRS=site -- OK
Running make for S/ST/STEVEB/WiringPi-API-2.36.4.tar.gz
cp lib/WiringPi/API.pm blib/lib/WiringPi/API.pm
Running Mkbootstrap for WiringPi::API ()
chmod 644 API.bs
/usr/bin/perl /usr/share/perl/5.20/ExtUtils/xsubpp  -typemap /usr/share/perl/5.20/ExtUtils/typemap -typemap typemap  API.xs > API.xsc && mv API.xsc API.c
Please specify prototyping behavior for API.xs (see perlxs manual)
cc -c  -I. -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"2.36.4\" -DXS_VERSION=\"2.36.4\" -fPIC "-I/usr/lib/arm-linux-gnueabihf/perl/5.20/CORE"   API.c
API.xs:8:20: fatal error: INLINE.h: No such file or directory
 #include "INLINE.h"
                    ^
compilation terminated.
Makefile:341: recipe for target 'API.o' failed
make: *** [API.o] Error 1
  STEVEB/WiringPi-API-2.36.4.tar.gz
  /usr/bin/make -- NOT OK
```

stevieb9, 2017-02-13:
Whoops! I added the INLINE.h include as I'm using some of its functionality, but it appears that I forgot to include it in the distribution!

Thanks for the heads-up. I'll get this fixed shortly.

commit 0046926c (2017-02-13): closes #8; added INLINE.h to manifest
closed 2017-02-13 by stevieb9