stevieb9/wiringpi-api #26: Docs: lcd_char_def() takes a string not an array as argument

gflohr, 2017-03-02:
The docs for lcd_char_def() say that the "data array is 8 bytes".  As a Perl programmer it took me some debugging until I found out that the data argument is actually a string, not a Perl array.

Maybe, the formulation in the docs should stress that a little more.  And it would be nice to have if the functions would check the type of argument and silently convert a Perl array into a string.

The same applies to the documentation of RPi::WiringPi::LCD::char_def().  There, the reference to the function lcdPutchar() should be replaced to the method print_char().

stevieb9, 2017-03-02:
The docs in that module as you've found are likely not accurate. Some of
the functions in LCD I wrapped I didn't get too much into detail with.

That said, I'll plug in one of my units, and review the whole thing. I may
have to go as far as accepting something Perl to `unsigned char*` like I
did in WiringPi::API's [SPI RW functionality](
https://metacpan.org/source/STEVEB/WiringPi-API-2.36.6/API.xs#L90).

Full review coming up over the coming week against the entire suite. POD
fix sweep at minimum, bug fixes the desire.

stevieb9, 2017-03-03:
Example [char defs](https://www.quinapalus.com/hd44780udg.html) for testing a potential fix.

stevieb9, 2017-03-04:
For now, I am going to rename the broken C functions (eg: lcdDefChar() for lcdCharDef()), until I get a better handle on how to do some needed trickery in the XS. This is brokenness, so getting it fixed is my main priority. Allowing the import of the original C function name will come after I've figured out said trickery.

I'm getting the fixes in now along with relevant doc fixes for review, and then we'll go from there.

commit d74efe66 (2017-03-10): closes #26; lcd_char_def() not setting char properly
closed 2017-03-10 by stevieb9