stevieb9/wiringpi-api #33: WiringPi-API failing all automated tests; needs to fail gracefully

jkeenan, 2018-03-22:
As can be seen from http://fast-matrix.cpantesters.org/?dist=WiringPi-API, the WiringPi-API is failing in all automated CPAN testing environments.  Here's what I got today when attempting to install it against perl-5.27.10 on FreeBSD-11.0, using cpanm and clang 3.8.0:
```
{
  author => "STEVEB",
  dist => "WiringPi-API",
  distname => "WiringPi-API-2.3614",
  distversion => 2.3614,
  grade => "FAIL",
  prereqs => undef,
  test_output => [
    "Building and testing WiringPi-API-2.3614",
    "cp lib/WiringPi/API.pm blib/lib/WiringPi/API.pm",
    "Running Mkbootstrap for API ()",
    "chmod 644 \"API.bs\"",
    "\"/usr/home/jkeenan/prep528/testing/perl-5.27.10/bin/perl\" -MExtUtils::Command::MM -e 'cp_nonempty' -- API.bs blib/arch/auto/WiringPi/API/API.bs 644",
    "\"/usr/home/jkeenan/prep528/testing/perl-5.27.10/bin/perl\" \"/home/jkeenan/prep528/testing/perl-5.27.10/lib/5.27.10/ExtUtils/xsubpp\"  -typemap '/home/jkeenan/prep528/testing/perl-5.27.10/lib/5.27.10/ExtUtils/typemap' -typemap '/usr/home/jkeenan/prep528/testing/perl-5.27.10/.cpanm/work/1521735236.757/WiringPi-API-2.3614/typemap'  API.xs > API.xsc",
    "mv API.xsc API.c",
    "cc -c  -I.  -DHAS_FPSETMASK -DHAS_FLOATINGPOINT_H -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_FORTIFY_SOURCE=2 -O2 -pipe -fstack-protector -fno-strict-aliasing    -DVERSION=\\\"2.3614\\\"  -DXS_VERSION=\\\"2.3614\\\" -DPIC -fPIC \"-I/home/jkeenan/prep528/testing/perl-5.27.10/lib/5.27.10/amd64-freebsd-thread-multi/CORE\"   API.c",
    "API.xs:22:10: fatal error: 'wiringPi.h' file not found",
    "#include <wiringPi.h>",
    "         ^",
    "1 error generated.",
    "*** Error code 1",
    "",
    "Stop.",
    "make: stopped in testing/perl-5.27.10/.cpanm/work/1521735236.757/WiringPi-API-2.3614",
    "-> FAIL Installing WiringPi::API failed. See /testing/perl-5.27.10/.cpanm/work/1521735236.757/build.log for details. Retry with --force to force install it.",
  ],
  via => "App::cpanminus::reporter 0.17 (1.7043)",
}
```
My hunch is that the reason for these failures is that you are not testing for some underlying library when you run ```perl Makefile.PL```.  Your README states:
```
This is an XS-based module, and requires wiringPi <http://wiringpi.com> version 2.36+ to be installed. The wiringPiDev shared library is also required (for the LCD functionality), but it's installed by default with wiringPi.
````
However, this requirement is not being enforced by your ```Makefile.PL```.  I'm not an expert on Perl modules built on top of non-standard C libraries, so I can't claim expertise here.  But I believe that modern versions of ExtUtils::MakeMaker can write a ```Makefile.PL``` which will provide a ```configure_requires``` metadata element that will enable a more graceful exit from compilation before ```make``` encounters the missing libraries.

Other CPAN distributions depend on this one.  They cannot be tested in automated testing environments until this problem is resolved.

Please let me know if I can be of further assistance.

Thank you very much.
Jim Keenan

stevieb9, 2018-03-22:
Thank you James for the report.

I could have sworn I put a check in there. I suppose not (or if I did, the
check is broken).

This is something I can get fixed up by tomorrow, and I will advise when
it's done (I'll test it on a fresh perlbrew 5.27.10 without the C library
installed).

Cheers,

Steve

stevieb9, 2018-03-22:
In fact, what I'll do is exit the build very early on if a specific
environment variable isn't set which is what I do elsewhere:

RPI_BOARD=1

One way or another, I'll fix it up.

jkeenan, 2018-03-22:
> Thank you James for the report.
> 
> I could have sworn I put a check in there. I suppose not (or if I did, the
> check is broken).

I see a LIBS element, but for some reason it's not DWImming.

You might want to explore use of 
http://search.cpan.org/~mattn/Devel-CheckLib-1.11/.  ISTR that if you 
use it right near the start of your Makefile.PL, you get a much more 
graceful exit.

There's also http://search.cpan.org/~dcantrell/Devel-CheckOS-1.81/ ... 
though I don't see a subclass for Raspberry Pi.

Thanks for your quick response.

stevieb9, 2018-03-22:
Thanks for that. I will investigate and go with a solution that works
across all Perlbrew instances that the distribution supports.

stevieb9, 2018-03-22:
I've got an automated build of 5.27.10 happening on three of my development
Raspberry Pis, as well as two of my Linux VMs currently. Once done, I will
get this fixed. Hopefully within a couple of hours.

commit 0cfefa73 (2018-03-23): closes #33; closes #32

33: We now gracefully exit from the Makefile.PL if wiringPi.h not found
32: EXPORT_TAGS POD did not contain all three tags
closed 2018-03-23 by stevieb9